FrameworkComputer/EmbeddedController #2: Building on Fedora

junaruga, 2022-01-21:
On Fedora 35, I was able to build like this.

With the compiler arm-none-eabi-gcc.

```
$ sudo dnf install arm-none-eabi-gcc-cs libftdi-devel

$ which arm-none-eabi-gcc
/bin/arm-none-eabi-gcc

$ make BOARD=hx20 CROSS_COMPILE=arm-none-eabi-
```

With the compiler arm-linux-gnu-gcc.

```
$ sudo dnf install gcc-arm-linux-gnu libftdi-devel

$ which arm-linux-gnu-gcc
/bin/arm-linux-gnu-gcc

$ make BOARD=hx20 CROSS_COMPILE=arm-linux-gnu-
```

If you don't mind, I would like to add the steps using `arm-none-eabi-gcc` on `README.md`.

kiram9, 2022-01-24:
@junaruga I tried this a while ago on fedora, and there is something with the build tools using arm-none-eabi-gcc-cs on fedora 35 which causes the application built with this to run incorrectly on the EC. For now I would not suggest using this to build EC binaries. 
I have not tried it yet, but I wanted to see if the zephyr tool chain worked, as I would prefer to migrate to this since most chrome-ec is moving to zephyr.

junaruga, 2022-01-24:
> I tried this a while ago on fedora, and there is something with the build tools using arm-none-eabi-gcc-cs on fedora 35 which causes the application built with this to run incorrectly on the EC

@kiram9 If you remember, could you tell me more about "there is something" and " run incorrectly on the EC"? What incorrect behavior happened? Is it reproducible?

junaruga, 2022-01-24:
> I have not tried it yet, but I wanted to see if the zephyr tool chain worked, as I would prefer to migrate to this since most chrome-ec is moving to zephyr.

I see. The zephyr tool chain is this? https://github.com/FrameworkComputer/zephyr

kiram9, 2022-01-24:
@junaruga It was reproducible, but will cause your laptop to not boot. 
It will show up as a WDT reset over and over, I think due to i2c timeouts. I think there is a mutex or something that is not updated causing the i2c event notifications to not notify the associated tasks, and when there are i2c transactions on the hooks task, it hangs, (hooks also does WDT patting). I have not had much time to debug what is going on yet.

The Zephyr sdk is https://github.com/zephyrproject-rtos/sdk-ng/releases and includes gcc for arm. 

The framework repo is for zephyr rtos, which we are not currently using for this project.

PR #3 (2023-05-26): Add a way to install development tools on fedora and arch linux to readme.
This PR fixes #2 by adding the way to install development tools on Fedora. I tested it on Fedora Linux 35. You can check the modified readme [here](https://github.com/junaruga/EmbeddedController/blob/wip/readme-fedora-build/README.md).