FrameworkComputer/EmbeddedController #11: Option to disable power LED entirely

hauntingEcho, 2022-03-07:
When playing music/youtube/whatever with the screen off as I go to sleep, the power LED lights up the whole room.  Options to dim it were added [here](https://github.com/FrameworkComputer/EmbeddedController/commit/6833f5e098e3198883693c73e527c3525f0cd1e5), but the ability to disable it entirely would be appreciated.  This has also been mentioned in the [discussion thread](https://community.frame.work/t/power-button-light/4297/16).

Putting electrical tape over a button gets a bit messier than putting it over the charge lights, and I haven't got up the courage to start cutting on the board just yet :)

hauntingEcho, 2022-03-07:
Exposing it via `/sys/class/leds` could also be nice (because we could set up keybinds), but the possibility of that is deeper than my firmware knowledge goes.

james2432, 2023-09-02:
I wish this was merged :(

hauntingEcho, 2024-10-21:
Looks like `/sys/class/leds` support may be available with kernel 6.11: https://old.reddit.com/r/framework/comments/1g8gbm2/what/lt0fxia/

AkechiShiro, 2024-10-22:
@hauntingEcho can confirm this works on kernel 6.11

I can shutdown the power LED with 
```
echo 0 | sudo tee /sys/class/leds/chromeos:white:power/brightness
```
Full brightness
```
echo 3 | sudo tee /sys/class/leds/chromeos:white:power/brightness
```

Here are also the other leds : 
```
chromeos::kbd_backlight chromeos:multicolor:charging chromeos:white:power input0::capslock input0::numlock input0::scrolllock
```

JohnAZoidberg, 2025-03-27:
See https://github.com/FrameworkComputer/framework-system/issues/37
We won't add the option to turn off the power button completely, as the LED is used to tell you the power state of the system.

But we are going to add an ultra low power mode and let you set the power button to any percentage you like, except 0%.

kariudo, 2025-05-11:
I still feel like there is value in having the option to set it to 0%, the LED has never worked on my FW13 AMD350; and a replacement input cover is on the way, but I actually had no idea it even was supposed to light up until a friend of mine asked out I disabled it (since his was illuminated).

Once input cover, I am concerned that since I am used to having everything nice and dark when I use my laptop in the dark, I am likely to find the LED annoying.

I think the side LEDs could be used under the same argument of having a visual representation of if the laptop is asleep or off, but honestly it seems pretty irrelevant if you are a poweruser who could manage to use this tooling to set it to 0% anyway. If you are about to interact with the computer and its asleep or off, just pressing a key to wake will tell you if its off really fast when it wakes, and in either scenario pressing the power button would wake/boot the device. So is it really a "it has to be on", since like I said, mine has never worked and it has never really caused me a problem.

TL;DR: What is really the harm in letting powerusers disable it?

t-8ch, 2025-06-28:
On any recent Linux system you can manage the LED through the standard APIs.

```sh
$ cd /sys/class/leds/chromeos:white:power/ # specific name depends on the device
$ echo 0 > brightness # turn off
$ cp max_brightness > brightness # turn on, unfortunately EC does not support non-maximum brightness
$ echo chromeos-auto  > trigger # switch back to automatic management through embedded controller
```

kiram9, 2025-12-30:
@queesamor have you tried framework-tool? https://github.com/FrameworkComputer/framework-system/releases

JohnAZoidberg, 2025-12-30:
https://github.com/FrameworkComputer/framework-system/issues/37