FrameworkComputer/EmbeddedController #15: Fn key rollover can trigger incorrect break codes, resulting in infinite key repeat

DHowett, 2022-06-10:
The easiest way I've found to reproduce this is by:

1. Pressing <kbd>Vol Down</kbd>
2. Pressing <kbd>Vol Up</kbd>
3. Releasing <kbd>Vol Down</kbd>

If done properly, releasing <kbd>Vol Up</kbd> will result in a break code for <kbd>F3</kbd> being generated. This can also be accomplished with some combination of <kbd>Fn</kbd> and the arrow/page keys.

It appears that the state-tracking variables `keep_fn_*` aren't transitioning through their state graph correctly.

https://github.com/FrameworkComputer/EmbeddedController/blob/8109392adb17a2cd7d30a74eee8613a0cb72ee7d/board/hx20/keyboard_customization.c#L238-L240

| Step | Action           | Call                           | Final State
| -    | -                | -                              | -
| 1    | Press Vol Down   | `hotkey_F1_F12(&0x0006, 0, 1)` | `keep_fn_key_F1F12` == `1`<br>Set to 1 because the key was pressed
| 2    | Press Vol Up     | `hotkey_F1_F12(&0x0004, 0, 1)` | `keep_fn_key_F1F12` == `1`<br>Set to 1 because the key was pressed
| 3    | Release Vol Down | `hotkey_F1_F12(&0x0004, 0, 0)` | `keep_fn_key_F1F12` == `0`<br>Cleared because the key was released
| 4    | Release Vol Up   | `hotkey_F1_F12(&0x0004, 0, 0)` | `keep_fn_key_F1F12` == `0`<br>**We exit early (see code below) because this is `0`**

https://github.com/FrameworkComputer/EmbeddedController/blob/8109392adb17a2cd7d30a74eee8613a0cb72ee7d/board/hx20/keyboard_customization.c#L296-L297

kiram9, 2022-06-14:
@DHowett see https://github.com/FrameworkComputer/EmbeddedController/commit/fa8982056aa0ad6267029cf3ea20dc5c59cba282

DHowett, 2022-06-15:
Ah, that looks perfect!

DHowett, 2022-07-09:
Yep, confirming fixed as of 9450caa

closed 2022-07-09 by DHowett