FrameworkComputer/EmbeddedController #19: Possibly use a "baseboard" to unify hx20 and hx30

DHowett, 2022-07-21:
Depending on how similar the hx* boards are, it may be possible--and perhaps advantageous!--to unify their implementations as a baseboard (see the `baseboard` directory for other examples.)

That would make it easier to abstract, say, the input cover/key layout and PS/2 emulation bits and drive bug fixes to both platforms simultaneously. Similar thoughts about the UCSI implementation, if the hx20/30 share one of those.

DHowett, 2022-08-26:
Barring this, perhaps it would be nice if the `hx20` and `hx30` board directories weren't on separate branches? They don't necessarily need to be unified, but... as the potential author of a number of changes that should apply to both 11th and 12th gen boards, it complicates deployment and development to have them on separate branches.

JohnAZoidberg, 2023-06-09:
We have combined them with a baseboard. I extracted much of the duplicated code into there.
The next hx20 and hx30 releases will contain those changes.

closed 2023-06-14 by kiram9