FrameworkComputer/EmbeddedController #25: Build error on Ubuntu 22.04.2 LTS:  <ftdi.h> not found

arthursw, 2023-03-13:
I cloned the repo and ran `make utils .` ; but got:

```

EmbeddedController$ make utils .
make: pkg-config: Aucun fichier ou dossier de ce type                      // No files of this type
make: pkg-config: Aucun fichier ou dossier de ce type
  VERSION ec_version.h
  HOSTCC  util/ectool
  HOSTCC  util/lbplay
  HOSTCC  util/stm32mon
  HOSTCC  util/ec_sb_firmware_update
  HOSTCC  util/lbcc
  HOSTCC  util/ec_parse_panicinfo
  HOSTCC  util/cbi-util
  HOSTCC  util/uartupdatetool
  BUILDCC util/ec_uartd
util/ec_uartd.c:24:10: fatal error: ftdi.h: Aucun fichier ou dossier de ce type          // No file of this type
   24 | #include <ftdi.h>
      |          ^~~~~~~~
compilation terminated.
make: *** [Makefile.rules:614 : build/bds/util/ec_uartd] Erreur 1    // Error 1

```

pantsman0, 2023-03-14:
Hi @arthursw, Canonical package the headers for libraries separate from the shared library files. Try again with the package `libftdi1-dev` installed.

arthursw, 2023-03-14:
Ouch, sorry I didn't even read the Readme, I only read [this post](https://www.howett.net/posts/2021-12-framework-ec/#using-fw-ectool). Sorry again about that!

closed 2023-03-14 by arthursw