FrameworkComputer/EmbeddedController #33: Support for 13th Gen

mbainrot, 2023-07-27:
Hi 

Would it at all be possible for support for 13th gen intel to be added/documentation updated to provide the entry point for 13th gen?

Thanks!

DeflateAwning, 2023-09-05:
Seems weird this isn't done yet. Any updates on when this'll get done, Framework?

crc-32, 2023-10-10:
Upon initial investigation using `ectool version` it looks like its still `hx30` commit fd2a203:
```
crc32@crc32-framework:~/EmbeddedController$ sudo ./build/hx30/util/ectool version
RO version:    hx30_v0.0.1-fd2a203
RW version:    
Firmware copy: RO
Build info:    hx30_v0.0.1-fd2a203 2023-05-09 21:13:19 runner@fv-az252-642
Tool version:  v0.0.16767-f6d6b927ee 2023-05-26 00:03:59 crc32@crc32-framework
```
No idea why RW version is empty here

DHowett, 2023-10-10:
> No idea why RW version is empty here

The version host command reads one version from the running image and one version from flash. On hx20/30, flash is locked out due to a pin mux conflict, so the version that isn’t currently running can’t be read.

You can unlock flash with host command 3E01 parameter 0x03 (byte), and re-lock it with parameter 0x00 (byte). 🙂

crc-32, 2023-10-10:
Aha :+1: makes sense thanks! I was mistaken thinking it was only locking writes not inaccessible entirely.