FrameworkComputer/EmbeddedController #38: AMD laptops disrupt sleep on lid close and charger plugin

septatrix, 2023-12-18:
See the [relevant forum thread](https://community.frame.work/t/tracking-framework-amd-ryzen-7040-series-lid-wakeup-behavior-feedback/39128/40). It seem like [multiple wakeup events are sent](https://community.frame.work/t/tracking-framework-amd-ryzen-7040-series-lid-wakeup-behavior-feedback/39128/40), from the lid switch and the keyboard (at least for the lid closing, unsure about the charging thing). This seems to cause operating systems to cancel sleep, i.e. closing the lid of a sleeping laptop wakes it up and so does plugging in the charger. [Someone from the Framwork team already has a theory](https://community.frame.work/t/tracking-framework-amd-ryzen-7040-series-lid-wakeup-behavior-feedback/39128/44)  about the code which might case this issue:

https://github.com/FrameworkComputer/EmbeddedController/blob/833fc2f25b82e0786a1a59e8b46ec327a53e4cec/zephyr/program/lotus/azalea/src/power_sequence.c#L313

brlin-tw, 2024-04-07:
Here's [the Linux kernel workaround patch that disables the keyboard wakeup source completely on firmware version 3.03](https://lore.kernel.org/platform-driver-x86/20231212045006.97581-1-mario.limonciello@amd.com/) that is included in >=6.7-rc7

xeruf, 2025-02-12:
that sounds like an issue I have been having - my battery really drains nicely on my AMD 16 on some days

JohnAZoidberg, 2025-03-27:
> that sounds like an issue I have been having - my battery really drains nicely on my AMD 16 on some days

Do you mean your system wakes up by itself? If no, then that's not the same issue.