FrameworkComputer/EmbeddedController #40: Swapping CTRL + FN

obsoleszenz, 2024-05-08:
I'm currently trying to swap the fn and ctrl keys on my framework 13 amd. I was able to map FN to CTRL with
`ectool raw 0x3E0C d1,d1,b2,b2,w14` but wasnt able to remap ctrl to fn. Is there any trick?

DHowett, 2024-05-08:
> wasnt able to remap ctrl to fn

What did you try?

obsoleszenz, 2024-05-08:
> > wasnt able to remap ctrl to fn
> 
> What did you try?

Looked for the SCANCODE_FN which is 0x0ff so i tried `ectool raw 0x3E0C d1,d1,b2,b2,w255` which didnt work though. But i just found out that my uefi has an option to swap fn and ctrl, so that works. But am wondering if theres an ec command to do this?

DHowett, 2024-05-08:
The thing about host command 3E0C is that it requires a _key matrix position_ and a _new scancode_. Since you are targeting a different key (the one labelled <kbd>ctrl</kbd>), you will need to update the two parameters here:

```
                         .--.--- here
                         v  v
ectool raw 0x3e0c d1,d1,b2,b2,wFF
```

...to match the row and column (in that order, in hex!) of the key labelled <kbd>ctrl</kbd>. You can figure out where _that_ is [here](https://www.howett.net/data/framework_matrix).

Instead of "2, 2" that would be "1, c" (row 1, column 12).

```
ectool raw 0x3e0c d1,d1,b1,bc,wff
```

But in this case, you should just use the firmware option for it. :)

obsoleszenz, 2024-05-08:
> The thing about host command 3E0C is that it requires a _key matrix position_ and a _new scancode_. Since you are targeting a different key (the one labelled ctrl), you will need to update the two parameters here:
> 
> ```
>                          .--.--- here
>                          v  v
> ectool raw 0x3e0c d1,d1,b2,b2,wFF
> ```
> 
> ...to match the row and column (in that order, in hex!) of the key labelled ctrl. You can figure out where _that_ is [here](https://www.howett.net/data/framework_matrix).
> 
> Instead of "2, 2" that would be "1, c" (row 1, column 12).
> 
> ```
> ectool raw 0x3e0c d1,d1,b1,bc,wff
> ```
> 
> But in this case, you should just use the firmware option for it. :)

Thank you :)

closed 2024-05-08 by obsoleszenz