FrameworkComputer/EmbeddedController #45: Add keyboard shortcut to toggle touch pad

bryanparadis, 2024-10-09:
Would be a nice addition that is available on most other laptops

JohnAZoidberg, 2025-03-27:
On Framework 13 and Framework 16, the touchpad is not controlled by the embedded controller, it's connected to and controlled by the CPU/OS. So I suggest you create a shortcut in your OS to do that.

On Framework 12 there will be a pin to the EC to disable the touchpad, so it would be possible to implement this.

JohnAZoidberg, 2025-05-03:
> On Framework 12 there will be a pin to the EC to disable the touchpad, so it would be possible to implement this.

I can't promise that we would add it, but if you could choose, which shortcut would you like to use?

bryanparadis, 2025-05-04:
It's nice they've added a pin on the EC in the 12. Imo it should be a first class feature on laptops.  I only have a 13 but the f keys sure are pretty full looking on the 12.

JohnAZoidberg, 2025-05-04:
> It's nice they've added a pin on the EC in the 12. Imo it should be a first class feature on laptops.

What scenarios would you use it for?
What's the reason you want to disable the touchpad and when?

> I only have a 13 but the f keys sure are pretty full looking on the 12.

Function keys are the same on all Framework 12, 13, 16

bryanparadis, 2025-05-06:
Sometimes you just do not want to fight with palm rejection, or you want to disable when you have another input device attached. It's been a feature on most of the laptops I have owned prior.