FrameworkComputer/EmbeddedController #46: Does this repo contain up to date code for each mainboard model?

imax9000, 2024-10-12:
Looks like there almost no activity here, and no clear instructions on how to build an image for each of released mainboard models.

The reason I'm curious is that supporting MagSafe3 cables requires a slightly modified USB PD negotiation protocol ([thread](https://community.frame.work/t/interested-in-a-magsafe3-expansion-card/34780/20)), meaning that one of the options is to patch EC firmware.

DHowett, 2024-10-12:
FWIW, I believe that you would need to reflash the PD controllers and not necessarily the EC.

This repository contains up-to-date source for each program, yes.

Here are the branches:

- `fwk-marigold-22606` - Intel Core Ultra 1 13
- `lotus-zephyr` - AMD 7xxx, both sizes
- `hx20-hx30` - Intel 11th, 12th and 13th Gen

imax9000, 2024-10-12:
WDYM by "PD controller" specifically?

DHowett, 2024-10-12:
There are two Cypress/Infineon CCG5 chips in each laptop[^1] in charge of USB C PD negotiation. They have their own independent firmware. I do not know if they are also the type-c port *controllers* or *managers*.

It is my understanding that the EC is not in charge of type-C negotiation in these devices.

[^1]: Well, I can't speak to the Laptop 16, but I can for the 13

imax9000, 2024-10-12:
Do you have any details on how does EC interact with them?

imax9000, 2024-10-12:
(also, can you please follow `@bsky.brid.gy@bsky.brid.gy` from your mastodon account, so I can follow you from Bluesky?)

imax9000, 2024-10-13:
https://github.com/FrameworkComputer/EmbeddedController/blob/lotus-zephyr/zephyr/program/lotus/src/cypress_pd_common.c looks like a good starting point. Sadly, public documentation on CYPD8225 leaves a lot to be desired, and it'll take a while for me to figure out if it runs `SinkPPSPeriodicTimer` on its own and if it does - how to configure it

closed 2024-10-16 by imax9000