FrameworkComputer/EmbeddedController #61: 4000 RPM Fan limit on FW16?

ltcptgeneral, 2025-03-02:
Hello. While using ectool, I noticed that setting the fan's RPM using `pwmsetfanrpm` did not match the reported RPM from `pwmgetfanrpm`. 

[image]

Trying speeds below 4000 RPM, it does match. It seems that there is a maximum fan RPM allowed in `pwmsetfanrpm`. However, by using `fanduty 100`, the fan reaches its maximum of ~5500 RPM.  

The 4000 RPM limit also seems to be the maximum that the default fan curve will reach. Why is the limit in place, and is it possible to increase this limit?

ltcptgeneral, 2025-03-02:
I think i've narrowed it down to `EmbeddedController/zephyr/program/framework/lotus/fan.dtsi`. I guess in order to change this I have to rebbuild and reflash. It would be nice if this was changeable from ectool directly.

JohnAZoidberg, 2025-05-03:
> The 4000 RPM limit also seems to be the maximum that the default fan curve will reach. Why is the limit in place, and is it possible to increase this limit?

That's right - it's the trade-off we chose where spinning the fan much higher increases the noise but doesn't give much more cooling performance. It's a subjective trade-off that we chose as a balanced max fan speed.

> I guess in order to change this I have to rebbuild and reflash. It would be nice if this was changeable from ectool directly.

As you found, `fanduty 100` works to get the fan to spin as fast as it can.
But keep in mind that during our validation we don't run the fans that fast.

Setting RPM is an inexact procedure - EC spins the fan via PWM and then measures how fast it's actually going and adjusts it little by little until it's running the requested RPM. It won't work well if you try to set it at a speed close to what the fan can physically handle.

closed 2025-05-03 by JohnAZoidberg