FrameworkComputer/EmbeddedController #62: [Feature Request]: Add an option to delay power button press

TheBlueKingLP, 2025-03-23:
### The Problem
I sometimes presses the power button accidentally(putting a bit too much pressure on it and triggered the button under the sensor) when my intention was to use the fingerprint reader.

### Proposed solution
A delay that requires the user to hold the power button for maybe 0.5 second to 1 second, would be appreciated. This is the behavior on some of my other laptops that has the fingerprint sensor on the power button.

JohnAZoidberg, 2025-03-27:
When the fingerprint reader is in use, i.e. when OS requests authentication, the power button functionality is already disabled.
There is no need for the workaround you describe.

Could you describe the exact situation that you are having issues with?

JohnAZoidberg, 2025-03-27:
Which device are you using? Maybe there's a bug on the particular system that you are using?
Here you can see the code that is responsible for this on the Intel Core Ultra Series 1 system: https://github.com/FrameworkComputer/EmbeddedController/blob/74a8e9ae2e2576e0e24e19cf49666708aedadea8/zephyr/program/framework/src/board_function.c#L116-L132

JohnAZoidberg, 2025-03-27:
https://github.com/user-attachments/assets/dac8e20d-dc5a-4f2e-8201-10979525fc22

TheBlueKingLP, 2025-03-27:
> Here you can see me trying it. The power button does not react, while the system is requesting fingerprint authentication.
> PXL_20250327_113219782.LS.mp4

Interesting, that is not the case for me. I just tried the `fprintd-verify` command as well, however unlike in your case, my Framework laptop went into sleep mode.
My unit is a Framework 13 AMD.

TheBlueKingLP, 2025-03-27:
In this case, I assume this would be a bug instead of a feature request then?

JohnAZoidberg, 2025-03-27:
> In this case, I assume this would be a bug instead of a feature request then?

That's right, will try to reproduce on AMD Ryzen 7040.

JohnAZoidberg, 2025-05-03:
Sorry, I forgot to answer, @TheBlueKingLP . I cannot reproduce your issue on the Ryzen 7040 systems either.

JohnAZoidberg, 2025-05-03:
Maybe you keep pressing the button after fingerprint verification has passed?
Could you send a video of what's happening?