FrameworkComputer/EmbeddedController #63: Some boards have a sing capacitor problem with mode == IDLE

jcdutton, 2025-04-21:
Hi,

I have read a comment that I think you put in the Embedded Controller source code.

"* Some boards have a sing capacitor problem with mode == IDLE. For such
* boards, a host can specify EC_CHARGE_CONTROL_FLAG_NO_IDLE, which
* makes the sustainer use DISCHARGE instead of IDLE. This is done by
* setting lower != upper in V2, which doesn't support the flag."

Please can you clarify this statement a little:
1) "Some boards", please can you tell me which boards.
2) "sing capacity problem".  What does that mean, what effect does it have?
Please can you describe what happens if one sets IDLE on those main-boards?

I have been trying to make some of my own changes to the EC source code, in order to improve the battery charging behavior and avoid mini-cycles of charge/discharge.
You can see my changes so far here:
https://github.com/jcdutton/EmbeddedController   (charge controller changes)
https://github.com/jcdutton/zephyr   (general improvements to port80
 handling)

I have a FW16 AMD, that does not appear to have the IDLE problem you mentioned above.
So, I am curious which boards have the problem?

jcdutton, 2025-04-23:
Ok, so I can safely assume that all FW13 Intel and AMD, and FW16 AMD have a working IDLE charger mode, and the "* Some boards have a sing capacitor problem with mode == IDLE"  does not apply to FW laptops.

JohnAZoidberg, 2025-04-23:
Yes, that comment is not from us but Google: https://github.com/FrameworkComputer/EmbeddedController/commit/d4f844ed162025e268884b015357e1e431187625
We have not seen the issue they mention on our systems.

jcdutton, 2025-04-23:
I have done some changes to the EC source code for lotus.
You can find them here:
https://github.com/jcdutton/EmbeddedController/commits/fwk-lotus-azalea-19573-jcd2/

I have also updated an "ectool" that controls and views its new functions.
https://github.com/jcdutton/ectool/commits/main/

I think it does a good job of fixing various battery charging related issue.
Please consider including some or all of them into the official FW 16 EC firmware.