FrameworkComputer/EmbeddedController #66: EC for AMD framework?

Kochanac, 2025-06-09:
Hello!

I see in the readme file that only intel core 11-13th gen are listed (hx20, hx30). I would like to flash an updated firmware on an AMD ryzen framework, can you tell me whitch one of them is it? hx30?

amstan, 2025-06-09:
Which AMD ryzen?

* fwk-lotus-azalea if it's 7040
* fwk-lilac if it's the "AI 300" released this year.

JohnAZoidberg, 2025-06-10:
> I would like to flash an updated firmware on an AMD 

This repository is not for getting your system updated, we provide the sourcecode for reference or experimenting with it.
To update your BIOS please get the official releases:

- https://knowledgebase.frame.work/framework-laptop-16-bios-and-driver-releases-BkeqkVovp
- https://knowledgebase.frame.work/en_us/framework-laptop-13-bios-and-driver-releases-amd-ryzen-ai-300-series-r1wqKAs1e
- https://knowledgebase.frame.work/framework-laptop-13-bios-and-driver-releases-amd-ryzen-7040-series-r1rXGVL16

rs-gh-asdf, 2025-06-18:
Hello,

I would also like to be able to remap the keys for my framework 13 7040. However the readme only has information about the Intel boards.

closed 2025-07-06 by Kochanac

Kochanac, 2025-07-06:
After digging around I also found this option for remapping keys: https://www.howett.net/posts/2021-12-framework-ec/#3e0c---keyboard-mapping

@rs-gh-asdf take a look, maybe it would help you. it seems flashing is not neccessary